ctranstrum/lintilla #2: Battery Life and Alternative Battery Options

kovasap, 2025-01-22:
I'm curious what battery life you usually get with your 250mah batteries.  The description says "months" but it's unclear to me if that involves a separate "dongle" battery pack.  Also, how long does it take to fully recharge them?

I ask because I'm considering just buying https://typeractive.xyz/products/lithium-battery-110mah, as they are cheaper than alternatives (given I'll be already ordering from typeractive and bundling shipping).  I'd like to estimate how much life I'll get on these in comparison to the 250mah batteries in the bill of materials.

ctranstrum, 2025-01-22:
I built two Lintillas about two month ago, and popped the batteries in at that time. Other than very short periods where I connected the keyboards to the USB cable long enough to flash the firmware on them, I have never fully charged any of the four halves. I have a dongle set up on one of the two pairs, and I use one set at work and the other at home, so potentially I use them less than if I just had the one keyboard. But for my dongle set, one half is still showing at just under 40% and the other at 70%, and for my non-dongle set, the master is reporting about 45%.

I would think you could get a fair amount of use out of a 110 mah battery. The best (but still imperfect) estimator can be found at https://zmk.dev/power-profiler and with 110mah, it estimates about a month on a non-dongled central and about three months on a peripheral.

closed 2025-01-22 by ctranstrum

kovasap, 2025-01-22:
Thanks for the info!  After doing some reading I finally now understand what a dongle actually is 😅 .  Did you follow a particular build guide to create your dongle?

ctranstrum, 2025-01-22:
Not exactly a build guide, no, but the fancy Prospector Dongle is here: https://github.com/carrefinho/prospector and the more traditional OLED dongle here: https://github.com/englmaxi/zmk-dongle-display. I've used both of those, but to be honest I've found that the rgbled-widget the Lintilla uses is, for my needs at least, almost as useful as the dongles, and I find myself preferring to run in it dongle-less.