ctranstrum/lintilla #3: Changing tolerance on case?

kovasap, 2025-02-08:
I just finished soldering my lintilla, and went to place it into my 3d printed case.  Sadly it does not fit neatly because (I'm pretty sure) that some of my hotswap socket soldering blobs are too big.  I was concerned to make them super strong when soldering, but think I overdid it.

AFAICT, the easiest way to fix this issue is to re-3d print the case, but with larger tolerances around the hotswap soldered sections.  When you created the case files, did you use a particular method to define the hotswap section outlines and set tolerances I could emulate?

I could also try to grind down the case somehow, but figured others may run into this problem too and a laxer case doesn't seem to have any downsides?

[image]

[image]

[image]

ctranstrum, 2025-02-08:
Haha, yeah you really overdid it with those connections. :) I'm not sure I want to make the tolerances as big as you'd need, and there's really no way for me to know which of the solder blobs is the largest without doing a little measuring, but you're correct that it probably won't hurt too much to increase the sizes of the cutouts for the sockets.

The fastest way is for you to do it yourself. If you clone the repo, you can edit the `config.yaml` file to give yourself more room, then re-run ergogen, and look in the outputs/cases folder for the new stl files. The lines to look for to edit would be lines 401 and 450 for the room around the sockets, currently set to half a millimeter of tolerance all the way around, then lines 783 and 791 for how high to make the hole, currently set to 2mm, though for the height, you can probably re-melt the solder to push it a little more flush, since the height of the hole is going to be more troublesome than the other setting due to the already thin nature of the case.

If I had to guestimate, I'd say to increase the `expand` settings to 1.5 and the `extrude` to 2.4.

It looks like the other components are probably okay, do you get the impression anything other than the sockets are having trouble fitting?

ctranstrum, 2025-02-08:
Ah, just re-read that the solder doesn't extend above the sockets. That's encouraging. I can give more room around the sockets, and I'll look at the other components as well while I'm in there.

ctranstrum, 2025-02-08:
Okay, I've just updated the case with a little more room around all the components. In particular, I bumped the space around each socket from .5mm to 1mm. I'd rather not go more than that because it makes the edge walls really thin, but you can always edit a copy yourself if you feel you need more and are okay with the thin walls that would result from that.

closed 2025-02-08 by ctranstrum

kovasap, 2025-02-08:
Amazing thank you!  I will try it out next week!

kovasap, 2025-02-11:
The new case works perfectly!  Thank you!!

ctranstrum, 2025-02-11:
So glad to hear that!