ctranstrum/lintilla #4: I put the xiao on the wrong side - desoldering ideas?

kovasap, 2025-02-25:
Hey i'm back again after finally getting my xiao microcontrollers!

I just went to finalized my build by soldering the xiaos and my batteries.  Unfortunately, I soldered them on the wrong side :(.  And I did both sides of the keyboard before double checking :(((

[image]

I managed to remove one by grilling it on a hot plate at 350F, but I'm pretty sure that destroyed my microcontroller:

[image]

A friend mentioned that the keyboard might still work with the controller on the wrong side (with some changes to the case), but I would rather just redo it correctly.  Do you happen to have any ideas as to how to remove the xiao without destroying it?

Worst case I can just grill the other side and get two new controllers.

Also sorry for all the github issues; if there's a better forum for this please let me know!

ctranstrum, 2025-02-26:
Ouch, yeah, my advice for removing an smd-soldered xiao would be to use hot air while the xiao is upside down so that as soon as the solder is liquid it will fall off; that would reduce the amount of heat to the absolute minimum. There's no guarantee, though, that the xiao would survive the ordeal, but that's probably the most likely to work.

Your friend is correct. Though the case would no longer fit without modification, electrically the xiao would still have all the correct connections on this board if installed on the wrong side.

It looks like removing the xiao also lifted some of the pads off the pcb, but that doesn't necessarily mean that the xiao is dead or that the pcb won't work with a working xiao soldered to the other side.

There is a dev board firmware you can flash to the xiao to see if it's working or not, though I don't have a link nor can I offer any help with that process. But if you want to try to google it, you may still be able to salvage one or more of the xiaos.  If the xiao is still working, you can use more flux and solder to remove the pcb's pads from it, and then solder it to the correct side. But as you said, a new xiao will be the more certain bet.

closed 2025-02-26 by kovasap

kovasap, 2025-02-27:
Today I tried the heat gun approach for the other xiao microcontroller and it worked really well!  I covered the controller with foil, put the whole board upside down, then heated one side at a time with the heat gun while pushing firmly from above with tweezers.  The first side came off cleanly, with the other side bending to allow separation.  Then the other side was easy to remove in the same way.  Any the xiao still works!

[image]

[image]

kovasap, 2025-02-27:
I've now assembled the entire keyboard and everything seems ok so far, although i still need to test all the switches.  Will send pictures of the final product!

ctranstrum, 2025-02-27:
Oh! That's a relief to hear! Nice job! Can't wait for final pics!