ctranstrum/lintilla #5: Viewing a wiring diagram

kovasap, 2025-02-27:
If it hasn't become abundantly clear by now, I'm still somewhat new to building keyboards 😅 .  My build is coming along nicely, but on one of my boards, two of the keys do not work:

[image]

I suspect that one of the connections to the microcontroller is not solid, or perhaps one of my diodes/sockets is not soldered properly?

What's the best way to view a wiring diagram of the board so that I can figure out which soldering jobs I need to check for these two keys?

ctranstrum, 2025-02-27:
One of the main weaknesses of the ergogen-based approach to creating keyboards is that it doesn't create a wiring diagram. All of the information for a wiring diagram is present in the `config.yaml` file, but not in a format as easily digestible as a wiring diagram. Those two keys are using P4 as the column pin and from top to bottom P3 and P2 for the row pins. Since they are both not working, my guess would be that you have a problem with the connection to P4 on the MCU.

ctranstrum, 2025-02-27:
One thing you can test is taking a pair of tweezers and touching P4 to P3 and then P4 to P2 to see if they trigger that way. If that works, then the MCU is still good, and you just need to find where it doesn't get back to P4 correctly. If shorting the pins doesn't work, then you may have a bad pin on the mcu.

kovasap, 2025-02-27:
Yep, my P4 was not soldered correctly.  I just added some more solder to that pin and it works now!

closed 2025-02-27 by kovasap

ctranstrum, 2025-02-28:
So glad to hear that! I'm really happy you didn't ruin either of your xiao's in the desoldering process. 😄

kovasap, 2025-02-28:
Haha well the one that i removed via frying pan was ruined, but I had a 3 pack :).

ctranstrum, 2025-02-28:
Ah, yeah, I was really surprised when I had thought the xiao lived through your first desoldering attempt. Well, thankfully you had the spare!