ctranstrum/lintilla #6: Case tolerance feedback

kovasap, 2025-02-27:
FEEL FREE TO IGNORE THIS FEEDBACK!  (as any open source feedback)  I intend it as just info, not as a request for help or something blocking my build that I need to figure out.

One thing I noticed with my build is that the upper inner corner of the case sandwich doesn't fit flush and can splay out.  For one of my sides, this is certainly because of sloppy soldering (and using a controller that was removed and reattached, with extra solder sticking around on it).  But it is present on the other side as well.  This issue is exacerbated by the absence of a screw on the inner side of the controller.

It may be worth considering giving a bit more vertical tolerance for the microcontroller where the soldering happens to reduce this effect.

[image]

ctranstrum, 2025-02-27:
This is something I found on my builds as well, though not quite as pronounced as you are seeing, and is definitely something that bugged me enough that I used a LOT more screws on my subsequent keyboard design.

[image]

If you don't mind removing the case and trying to get a good picture of how high the solder goes on the back of the mcu, that might help. While I can't add a screw to that location without making some major redesign choices, I can update the case files fairly easily, since they are auto-generated, and maybe we can add a bit more to the tolerances around the solder points of the mcu, like we did for the sockets. A side view picture would probably show it best.

kovasap, 2025-02-27:
Actually, upon further inspection, it looks like my pcb may be slightly warped, perhaps due to the hot air station use to get the xiao off.  Additionally, it seems like my xiao is just uniformly vertically shifted up when compared to the other side, likely because of the extra solder that was already on it from the first application.  

I'm not sure what would help for my situation in terms of case changes.  But I do think generally giving a tiny bit more vertical space for the xiao would make the case more fool-proof.

[image]

ctranstrum, 2025-02-28:
Yeah, it's hard to tell from the photo if the pcb is warped, but I trust you if you say it is. What I find interesting is that on my builds, it seems to fit just fine on the top of the pcb, where there is no solder, but the bottom case bends away at that corner, whereas on yours it's the top case that is also bending away, or is that the warp of the pcb I'm seeing?

I had prepared some changes to allow a little room for solder above the mcu (or, since it's soldered upside down maybe I should say below it), but that doesn't look likely to alleviate your issue at all now that I see what you are saying about the solder creating space between the mcu and the pcb. My case dimensions assume the mcu is installed flush to the pcb.

kovasap, 2025-02-28:
I think that is the pcb warp you are seeing.

Thanks!  Those changes may still be helpful in general.

I'm pretty ok with the current state of my build, this issue isn't very noticeable unless i'm looking for it.

ctranstrum, 2025-02-28:
Gotcha. Updated the cases with increased tolerance for small solder blobs above the microcontroller. I doubt it will help much with your issue, but I agree it's a good idea.

closed 2025-02-28 by ctranstrum