ctranstrum/lintilla #9: 3d printing case question

ntnrmr, 2025-04-08:
I came over this keyboard and think that I like the layout and number of keys. This will be my first build and I decided to use jlc services to print both the pcb and the case. While pcb has been approved, review of the case got some yellow alerts and I've received subsequent email from the service asking to adjust the model. Could you offer some advice on that matter? Are this errors safe to ignore?

Also I can see that in their model viewer, halves have some weirdly looking red areas which I suspect might be a bug of some sort

[image]

> the wall thickness of those indicated red area is too thin, which has high risks of loss, deformation ,crack, or damage during the printing process.
> 
> Could you please confirm if the risks are acceptable for you?
> 
> If not, could you please kindly increase it to be at least 0.8 mm to proceed? 1.5 mm will be better .Then we will activate the ' replace file' button for you.
> 
> Besides,the part has high risks of deformation and warping because of the geometry.Could you please confirm if the risks are acceptable for you?
> 
> Usually, for long parts, flat parts, frame parts, enclosures with no strong ribs inside, and large unoccupied areas parts, the models will meet potential risk to shrink to the center or warp to the diagonal lines and edges.

[image]

closed 2025-05-07 by ntnrmr

ctranstrum, 2025-05-08:
Thanks for the update. Sounds like you had similar results, though maybe slightly worse, to mine. What material did you choose to print?

PeterCLarsen, 2025-11-10:
I'd be interested to know if the worst component was your top right plate. I just loaded the files to PCBway for a quote for printing in petg.  It auto quoted for the other three files but told me to "Ring for Quote" for the right upper plate.  The error message was:

> Note: There is no system auto-quote since it failed to render the 3D file(Reason might be: 1. Some formats cannot be rendered, such as SLDPRT. 2. Network latency.）. Your file is correct and available, please submit the order and wait for the formal 

thinking this was a file corruption during download, I went back and downloaded it again, but received the same error.

I also see that this file was not updated in the repository at the same time as the other three.