ctranstrum/lintilla #10: Keymap json file?

behlersyeg, 2025-04-15:
I am trying to use the keymap editor gui (https://nickcoutsos.github.io/keymap-editor/) to edit the keymap as I am very much a zmk beginner and had previously been using VIAL to make changes on my Corne. 

When I try and link it to my repo, however, it says I need a json file. 

I tried loading in the following: 

```
{
  "layouts": {
    "LAYOUT": {
      "layout": [
        { "label": "LShift", "row": 0, "col":  1, "x":     0, "y": 2.360 },
        { "label":    "Esc", "row": 0, "col": 14, "x":    14, "y": 2.360 },
        
        { "label":    "Tab", "row": 1, "col":  0, "x":     0, "y":  0.360 },
        { "label":      "Q", "row": 1, "col":  1, "x":     1, "y":  0.360 },
        { "label":      "W", "row": 1, "col":  2, "x":     2, "y":  0.110 },
        { "label":      "E", "row": 1, "col":  3, "x":     3, "y": -0.015 },
        { "label":      "R", "row": 1, "col":  4, "x":     4, "y":  0.110 },
        { "label":      "T", "row": 1, "col":  5, "x":     5, "y":  0.235 },
        { "label":      "Y", "row": 1, "col":  10, "x":     9, "y":  0.235 },
        { "label":      "U", "row": 1, "col":  11, "x":    10, "y":  0.110 },
        { "label":      "I", "row": 1, "col":  12, "x":    11, "y": -0.015 },
        { "label":      "O", "row": 1, "col": 13, "x":    12, "y":  0.110 },
        { "label":      "P", "row": 1, "col": 14, "x":    13, "y":  0.360 },
        { "label":   "BSPC", "row": 1, "col": 15, "x":    14, "y":  0.360 },
      
        { "label":   "Ctrl", "row": 2, "col":  0, "x":     0, "y":  1.360 },
        { "label":      "A", "row": 2, "col":  1, "x":     1, "y":  1.360 },
        { "label":      "S", "row": 2, "col":  2, "x":     2, "y":  1.110 },
        { "label":      "D", "row": 2, "col":  3, "x":     3, "y":  0.985 },
        { "label":      "F", "row": 2, "col":  4, "x":     4, "y":  1.110 },
        { "label":      "G", "row": 2, "col":  5, "x":     5, "y":  1.235 },
        { "label":      "H", "row": 2, "col":  10, "x":     9, "y":  1.235 },
        { "label":      "J", "row": 2, "col":  11, "x":    10, "y":  1.110 },
        { "label":      "K", "row": 2, "col":  12, "x":    11, "y":  0.985 },
        { "label":      "L", "row": 2, "col": 13, "x":    12, "y":  1.110 },
        { "label":      ";", "row": 2, "col": 14, "x":    13, "y":  1.360 },
        { "label":      "'", "row": 2, "col": 15, "x":    14, "y":  1.360 },
      
        
        { "label":      "Z", "row": 3, "col":  1, "x":     1, "y": 2.360 },
        { "label":      "X", "row": 3, "col":  2, "x":     2, "y": 2.110 },
        { "label":      "C", "row": 3, "col":  3, "x":     3, "y": 1.985 },
        { "label":      "V", "row": 3, "col":  4, "x":     4, "y": 2.110 },
        { "label":      "B", "row": 3, "col":  5, "x":     5, "y": 2.235 },
        { "label":      "N", "row": 3, "col":  10, "x":     9, "y": 2.235 },
        { "label":      "M", "row": 3, "col":  11, "x":    10, "y": 2.110 },
        { "label":      ",", "row": 3, "col":  12, "x":    11, "y": 1.985 },
        { "label":      ".", "row": 3, "col": 13, "x":    12, "y": 2.110 },
        { "label":      "/", "row": 3, "col": 14, "x":    13, "y": 2.360 },
        
      
        { "label":    "GUI", "row": 4, "col":  5, "x":  3.48, "y":  3.13 },
        { "label":    "LWR", "row": 4, "col":  6, "x":  3.48, "y":  3.13, "r":  15, "rx": 3.98, "ry": 7.63 },
        { "label":    "SPC", "row": 4, "col":  7, "x":  3.48, "y":  2.63, "r":  30, "rx": 3.98, "ry": 7.63, "h": 1.5 },
        { "label":    "ENT", "row": 4, "col":  8, "x": 10.52, "y":  2.63, "r": -30, "rx": 11.02, "ry": 7.63, "h": 1.5 },
        { "label":    "RSE", "row": 4, "col":  9, "x": 10.52, "y":  3.13, "r": -15, "rx": 11.02, "ry": 7.63 },
        { "label":    "ALT", "row": 4, "col":  10, "x": 10.52, "y":  3.13 }
      ]
    }
  }
}
```

But the tool kicks out "Could not parse keymap
It looks like some bindings in your keymap are using custom pre-processor macros which isn't supported by the Keymap Editor. These macros make it difficult both to resolve bindings to behaviors and parameters, as well as to generate modified keymap code that is consistent with your original intentions."

Is there a way to get an official zmk json layout file? Here is the thread discussing it: https://github.com/nickcoutsos/keymap-editor/issues/293

ctranstrum, 2025-04-15:
Yeah, I had problems using Nick's keymap editor as well when I tried it a few years back, and I opted not to use it because it was opinionated about not allowing preprocessor macros. It looks like the issue you linked on his repo has resolved basically everything except the preprocessor macros. But if you have your own repo set up and the lintilla files copied over into it and have followed the notes in the linked issue, you can just use a search and replace in your favorite editor to remove the preprocessor macros, and I imagine it should then be usable in Nick's keymap editor.

behlersyeg, 2025-04-16:
Good point! I wasn't sure how critical the preprocessor macros were, but now that I think about it, I can always copy and paste them back in after generating a keymap. Thanks!

closed 2025-04-16 by behlersyeg