ctranstrum/lintilla #11: Help with firmware?

behlersyeg, 2025-04-18:
Hey, so I had a working version of the firmware, and was even able to edit the first layer to make it Colemak, but then somewhere along the way, the new firmware stopped working so I tried to start over and see if I could get back to a working version of the original keymap. 

No matter what I try, I can't get the two halves to pair. Dropping in older firmware, and the connect, no problem. 

Can you take a look at my repo and let me know what I missed? 

https://github.com/behlersyeg/zmk-config

ctranstrum, 2025-04-18:
So, a couple of things about zmk firmware to take into consideration.

1. Almost all connectivity problems are resolved by flashing a settings reset file to the boards and then starting again. https://zmk.dev/docs/troubleshooting/connection-issues
2. The zmk project is a live project and every once in a while they accidentally break things and newly compiled firmware won't work until they unbreak it. Luckily this is fairly rare and is usually fixed within a few days.
3. I actually made a few changes to the Lintilla firmware myself in the past few days and had one issue I wasn't sure what to do about, so maybe it was me that broke it. My issue was on dongle code, though, so I'm not sure it would affect you.

I can take a look, sure, but if the code is compiling and you are getting firmware out of it, I may not be the best person to help. You may have better luck talking to smarter people on the zmk discord server.

ctranstrum, 2025-04-18:
Looking at your repo, the current code looks to be compiling correctly, but it also looks not to have Colemak, so when you say going back to older code, do you mean a) reflashing a firmware you compiled a few weeks ago? or b) restoring files to their old version, recompiling, and then flashing that new firmware to the board?

behlersyeg, 2025-04-18:
I meant more option B). I nuked the old repo and started over. I knew it worked with the default keymap, so I am just retracing my steps. 

I'll try going through the zmk connection issues with a fine tooth comb again, I guess! Thanks for the tips and context and I'll keep trying! 

One small piece of feedback, it may be useful to have a more basic "default" keymap without all the heavy modifications. For example the default Corne is just three layers with basic &kp inputs and loads of white space. I personally find it a bit easier to customize something from the ground up, than tear stuff down and rebuild on top. But that's just a small thing that has more to do with the learning curve going from VIA to ZMK, hah hah.

ctranstrum, 2025-04-20:
How is the process going? Have you made any headway on this?

Thanks for the feedback. My standard layout uses like 9 layers and Colemak as the base keymap, so this was my attempt to simplify, LOL, but I'll take another look and see if I can pare it down even more.

ctranstrum, 2025-04-20:
Eh, I finally had a chance to look at my most recent firmware changes, and found a mistake that probably was the sole cause of your issues. I fixed it just moments ago, and now all my firmware is compiling successfully. I bet it will now work for you as well.

ctranstrum, 2025-06-03:
Closing this issue since there has been no activity on it for over a month, but feel free to open a new issue if you continue to have problems building the firmware.

closed 2025-06-03 by ctranstrum