ctranstrum/lintilla #12: Troubleshooting non-working switches

otherdave, 2025-04-21:
Hi! Thanks for this project. I saw you post in another issue with advice on the pins to check for some bad switches. Any chance you've got some more advice to spare?

The right-side works perfectly. The left side however....

[image]

All of the missing switches are ones that don't work.

(One of them was a badly soldered diode, but I forget which exactly)

I have shorted some of the pins on the controller together - I DO get keypresses there (I got the pins from the other issue you commented on since one of those keys overlapped with one of mine not working)

In case it's relevant, I realized today that my controller appears to always be on. If I turn the switch from off to on, nothing happens. If I plug in the usb-c cable, even when the switch is off, it turns on. I just realized that earlier today and haven't had a chance to check & re-solder the switch and check the controller. But I'm mentioning it here in case it's helpful.

My planned next steps are to remove & re-solder the switch and then re-re-re-check the controller and make sure everything is soldered correctly.

ctranstrum, 2025-04-21:
Regarding switches not working, the best troubleshooting guide I have seen is by splitkb. I don't think I can give better troubleshooting advice than what is already written so well there. https://docs.splitkb.com/troubleshooting/key

Unfortunately when you use ergogen to create the keyboard, it does not automatically create a wiring schematic, so if you need to know which pin connects to which switch, you will need to dig into the file named config.yaml to see if you can figure it out. I can help with that process if needed.

It's not entirely intuitive how the switch works. The switch only affects whether or not the battery and the controller are connected. So when the USB cable is unplugged, the switch determines whether the controller gets electricity and powers on. But when the USB cable is inserted, the controller will get power via USB and turn on regardless of the position of the switch. However, in that case the battery will not charge until the switch is moved to the on position.

otherdave, 2025-04-22:
Thanks so much for the advice / info! The `config.yaml` wasn't that scary after all, but it turns out I didn't need it. I re-soldered all of the connections to my controller and things just magically worked! I suspect the ground was a really bad joint and then there may have been a few others that just weren't making contact at all / consistently. So I gave them all a look & reflow or more solder. I touched up a few diodes that looked suspect just to be sure and things are great!

I'm glad I went through the yaml file, it was fun to learn a bit more about how this is put together.

Thanks again for the help and for Lintilla.

ctranstrum, 2025-04-22:
Glad to hear it! Marking this as closed.

closed 2025-04-22 by ctranstrum