ctranstrum/lintilla #13: Generating images of the keymap?

otherdave, 2025-04-22:
What tool did you use to generate the images of your keymap in the firmware repo?

I wanted to do the same and I've seen some tools (zmk-viewer) and zmk.studio but they don't quite as nice as yours.

ctranstrum, 2025-04-22:
It's called keymap-drawer. It's a tiny bit complicated to get set up so that it creates the keymap automatically, but you can check out the files in the zmk branch of my repo for it. The first is in `.github/workflows/draw.yml`, and the other in `keymap_drawer.config.yaml`. It's (probably?) easier just to use the website: https://keymap-drawer.streamlit.app/

otherdave, 2025-04-23:
Ah, thanks again! The website was perfect (once I figured out which .dtsi to upload so it had the layout of the keyboard).

closed 2025-04-24 by ctranstrum