ctranstrum/lintilla #17: Using a different power switch

sinefabula, 2025-05-12:
Hi! First of all, great case design! I think all popular keyboards should be upgraded to your case design :)

Anyway, I'm planning to build this keyboard but I can't order those power switches from typeractive (as they don't ship to my country in the EU) and ordering from Mouser (or similar) would cost a lot. Can the power switch be replaced with something else with the same footprint? Or I have to update the PCB and the case to make it fit? If so, do you by chance have any guidelines on how to modify these? From my understanding, you're using some JS library for parametrised modelling and PCB design? Also, this could be helpful for those wanting to adjust the position of the thumb cluster.

ctranstrum, 2025-05-12:
Thanks for the positive feedback. The reset buttons can be replaced with another momentary button that has the same footprint, sure. A lot of people shop on chinese websites that have very low prices for electronics components, though sometimes shipping can take longer. There's not really anything special with the switch I've recommended over any other that matches the dimensions.

As far as making changes to the footprints, this project is using ergogen to make everything relatively easy to adjust. While there is a learning curve involved with it, ergogen can allow you to update any part of the keyboard, from the reset button to the thumb cluster, as you have mentioned.

I should also point out that a reset switch in not strictly necessary for the keyboard. The reset button does two things: press it once to restart the firmware, or double-tap it to enter firmware flashing mode. You will need to double-click the reset button the first time you flash the firmware onto the board. You can do that directly on the xiao before the case is installed. After that, as long as you dedicate a key somewhere in your keymap that will allow the board to enter flashing mode, you will not need to use a reset button. If you need to reboot the firmware at any point, using the power switch to turn it off and then on again will restart the firmware.

ctranstrum, 2025-07-03:
Closing this issue, since there has been no activity on it for more than a month.

closed 2025-07-03 by ctranstrum