ctranstrum/lintilla #18: Footprint compable

thanhdat77, 2025-05-19:
Thanks for your work
Can i ask is it fit chocv2, i heard that the center footprint in v2 is a bit bigger

ctranstrum, 2025-05-19:
You heard correctly. Choc v2 require a larger center hole than choc v1. Unfortunately, I designed this only for choc v1. So no, this board is not compatible with choc v2 switches. Sorry.

thanhdat77, 2025-05-20:
Sorry to bother you, but I’ve already bought the Choc V2 switches and keycaps. I was wondering — do the keyboard have MX-compatible spacing? And if not, would it be possible to modify the plate by enlarging the holes?

ctranstrum, 2025-05-20:
It's no bother. The spacing between keys is not standard MX. Instead, it's halfway between standard MX and standard choc spacing. Since it doesn't take choc v2 switches, I've never actually tried it with MX keycaps, so I can't say for sure that it doesn't work, but I would be inclined to think that it won't fit.

My newest keyboard design uses MX spacing and accepts both choc v1 as well as choc v2 switches, and it has the same splay that the Lintilla has, but it's definitely not a Lintilla. If you want to take a look at it, you can find it at [github.com/ctranstrum/chuck](https://github.com/ctranstrum/chuck).

If you can't find a keyboard design you like other than the Lintilla, you can always edit the Lintilla files to modify the spacing and add the choc v2 footprints, but that is a fairly difficult project if you've never created your own keyboard before, so you may be better off, if you want to go that route, of designing your own board to your exact specifications.

closed 2025-07-03 by ctranstrum